zlatko-minev/pyEPR #5: Analyzis of a subset of modes

leghtas, 2017-11-01:
If I run the analysis on a number of modes which is smaller than the number of analyzed modes, I get an error:
```
  File "D:\Users\Zaki\github\pyEPR\pyEPR\core.py", line 1005, in analyze_variation
    CHI_O1= 0.25* Om * PJ * EJ.I * PJ.T * Om * 1000.             # MHz

  File "C:\ProgramData\Anaconda3\lib\site-packages\numpy\matrixlib\defmatrix.py", line 343, in __mul__
    return N.dot(self, asmatrix(other))

ValueError: shapes (2,2) and (1,1) not aligned: 2 (dim 1) != 1 (dim 0)
```

I see this is due to the fact that Om comes from all the computed frequencies, and not only the analyzed ones. 

Has this been solved in another fork / branch? if not I will fix it.

zlatko-minev, 2017-11-01:
Great call. No this has not been solved elsewhere. Glad to have your
solution :)

zlatko-minev, 2017-11-01:
Solved by Zaki

closed 2017-11-01 by zlatko-minev