zlatko-minev/pyEPR #21: incompatible with Pandas 0.25.1

calebjordan, 2020-01-28:
During do_EPR_analysis()

`AttributeError: module 'pandas.compat' has no attribute 'StringIO'`

https://stackoverflow.com/questions/58372475/attributeerror-module-pandas-compat-has-no-attribute-iteritems

Just a heads up!

zlatko-minev, 2020-01-28:
Oh! This is a change in the new version? Thanks for pointing this out. Do you have a solution?

calebjordan, 2020-01-28:
Downgrading to 0.24.2 worked for now.

zlatko-minev, 2020-01-28:
I updated the readme for now, thanks! Let me know if you run into other issues. The master branch will soon turn into an official v0.8 with a `pip install` and online docs.

DanielCohenHillel, 2020-05-05:
I've just made a commit to fix this, please check if it solves your problem😀

closed 2020-05-07 by DanielCohenHillel

zlatko-minev, 2020-05-07:
Nice, Pandas is now at v1 so this is good to close