zlatko-minev/pyEPR #25: error when calling get_Qdielectric directly

ptruitt, 2020-04-23:
If one wants to used DistributedAnalysis.get_Qdielectric (Q_surface, etc.), there is a problem in that self.U_E is not updated and might be from a different variation. Found when I ran do_EPR_analysis on several variations and then later tried to use get_Qdielectric on the first variation. Of course, self.U_E was still on the value from the last variation and I was getting a participation ratio above 1.

zlatko-minev, 2020-04-24:
Thank you, ptruitt (sorry, not sure who this is from the github name). 

Yes, you are correct. That code need to be updated. Do you want to create a pull request if you have fixed it? I am a bit swamped atm, so i wouldnt be able to get to it for some time right now.

zlatko-minev, 2020-05-05:
I think you fixed this, right? I guess we can close this issue now

closed 2020-05-05 by ptruitt