zlatko-minev/pyEPR #31: Create a development branch

DanielCohenHillel, 2020-05-06:
I have not been using this repository for that long but in the time I have,  I encountered problems several times where a code written just a few weeks ago didn't work correctly because of changes made to the repository. I suggest creating a "stable" branch (the `master`) and a `development` branch to solve this issue. 

This way you can still develop easily and when it's time to update the stable branch you can make sure that someone using an "old" method of doing things it would work but he would bet a "deprecated" warning or something.

zlatko-minev, 2020-05-06:
I agree, that’s how it should be.pyepr  was in dev for a long time, but now that’s it’s used by a lot of people it’s time to have a dev branch. Do you know how to add one?

DanielCohenHillel, 2020-05-06:
Yeah, I can create one.  But if so we need to make the master stable.

By the way yesterday I created a branch to fix a TODO from project_info.py (converting the self.dissipative from a class to a dictionary). You can check it out but it doesn't run now because I forgot some commas in the dictionary :\ I'll fix it today and then we can merge it back. I also created a TODO.md on that branch from the TODOs on all the files.

zlatko-minev, 2020-05-06:
Great job, i'll keep this open until this is all set

DanielCohenHillel, 2020-05-09:
I've created the a development branch and merged the branch `dissipative_class-to-dict` into it. I suggest following the git workflow described [here](https://nvie.com/posts/a-successful-git-branching-model/). There's a useful plugin you can use called git-flow which you can download from [here](https://github.com/nvie/gitflow/wiki/Installation). Also found [this](https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow) great tutorial you can check out.

zlatko-minev, 2020-05-09:
Yes, that’s great. I agree with the flow.

We can keep it simple enough so it’s easy and fast for us. 

I haven’t used the plugin. Does it just create all the types of branches. I use a GUI for the fit management, which seems to work well for git

DanielCohenHillel, 2020-05-09:
I'm not sure how would that plugin work with the GUI but it shouldn't really change much. All it does is give you a one-line command to do stuff that would normally take you several of commands to do (such as creating feature branches, merging branches correctly, etc...). It also automatically creates the branches but since I already used it to create them it shouldn't matter anymore.

I'm closing this issue. Hopefully we won't have any git workflow issues in the future 😀

closed 2020-05-09 by DanielCohenHillel

zlatko-minev, 2020-05-09:
Check out SourceTree by Atlassian, its good.  https://www.google.com/search?q=source+tree&rlz=1C5CHFA_enUS817US817&sxsrf=ALeKk01JHCO3XJk9Mhlzm_N0ozKssyMF7Q:1589063785311&source=lnms&tbm=isch&sa=X&ved=2ahUKEwj4oMvz66fpAhXBmHIEHZKwBVsQ_AUoAnoECBAQBA&biw=1536&bih=757#imgrc=5es7xfr_i3OuoM