zlatko-minev/pyEPR #32: Bug: Field calculator error when no variables in design

ptruitt, 2020-05-06:
I noticed that (at least some) field calculator operations fail when there are no variables in the design. One can get around this by just adding a variable to the design and setting it to a constant value if you don't want/need to vary it, but this isn't obvious at first. I have not found a fix for this. Sample error shown below:

com_error                                 Traceback (most recent call last)
<ipython-input-12-2a1ff56839e3> in <module>
----> 1 test.get_Qdielectric('Substrate', '0', '0')

~\code\pyEPR\pyEPR\core_distributed_analysis.py in get_Qdielectric(self, dielectric, mode, variation, U_E)
    840     def get_Qdielectric(self, dielectric, mode, variation, U_E = None):
    841         if U_E == None:
--> 842             U_E = self.calc_energy_electric(variation)
    843         Qdielectric = OrderedDict()
    844         print('Calculating Qdielectric_' + dielectric + ' for mode ' +

~\code\pyEPR\pyEPR\core_distributed_analysis.py in calc_energy_electric(self, variation, volume, smooth)
    595 
    596         lv = self._get_lv(variation)
--> 597         return A.evaluate(lv=lv)
    598 
    599     def calc_energy_magnetic(self,

~\code\pyEPR\pyEPR\ansys.py in evaluate(self, phase, lv, print_debug)
   2768             args.extend(["Freq:=", self.setup.solution_freq])
   2769 
-> 2770         self.calc_module.ClcEval(setup_name, args)
   2771         return float(self.calc_module.GetTopEntryValue(setup_name, args)[0])
   2772 

~\anaconda3\envs\pyEPR\lib\site-packages\win32com\client\dynamic.py in ClcEval(self, solnName, varVals)

com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024365), None)

zlatko-minev, 2020-05-06:
It's funny it fails on getQDielecitc. Id expect it to fail on a junction. It seems like mabe the issue is with the frequencies. did you run a simulation that now has results or was this before

ptruitt, 2020-05-06:
I think I found the issue. When lv = self._get_lv(variation) occurs and there are no variables in the design, you get lv = [''] (a list with one empty string), which causes the error. In the case of no design variables, we want it to set lv =[] (just an empty list). I confirmed that this fixes it on some small functions I wrote, but I'm not sure where/how to add a check for this in pyEPR to fix it wherever it might come up.

zlatko-minev, 2020-08-13:
@ptruitt   I’d like to invite you to get together for a Working Group Meeting for pyEPR.
 
See https://github.com/zlatko-minev/pyEPR/issues/45
 
Would you be interested?
 
Best wishes,
 
Zlatko

zlatko-minev, 2020-09-11:
> Hi Zlatko,
>
> Sure, this sounds like a good idea. I saw in the thread that you're
> thinking mid-late September, which should work for me.
>
> Best,
> Patrick

Hi Patrick,

 

The pyEPR working group meeting link is 

 

Meeting signup link: https://forms.gle/oYsCjCMZLZtfrGuFA

Discussion link: https://github.com/zlatko-minev/pyEPR/issues/45

 

Can you sign up / help spread the word to anyone you think would be interest. 

 

Thank you,

 

Zlatko

zlatko-minev, 2020-09-11:
We will have a working group meeting for pyEPR soon:

Meeting signup link: https://forms.gle/oYsCjCMZLZtfrGuFA
Discussion link: #45

You can you sign up / help spread the word to anyone you think would be interest.

zlatko-minev, 2026-05-17:
Still a real edge case. When a design has no variables, `_get_lv()` returns an empty list but `ClcEval` expects at least the frequency argument. The fix would be to handle the empty-variable case in `evaluate()` in `ansys.py` — skip the variable list when empty and only pass the frequency. Needs live Ansys testing to verify. Keeping open as a confirmed bug.

---
_Generated by [Claude Code](https://claude.ai/code)_