zlatko-minev/pyEPR #34: Remove qutip package

DanielCohenHillel, 2020-05-06:
Qutip is a great package and it provides many, very useful functionalities for quantum mechanics. One of the main things it can do is represent a quantum system with the `qobj` class. A  `qobj` object is not much more than a `numpy ndarray` but it has some problems. Doing operations on `qobj`s is much slower than simply working with a numpy array (I would say around ~15 times slower on average, obviously depending on what you're doing and how much), also it manages memory worse and uses system resources less effectively. To be fair, numpy is one of the most used python packages in the world so obviously it's super optimized.

Anyway, for pyEPR, qutip is used on around ~10 lines and everything it is used for can be done with a numpy array without really changing the code much. I got this idea when I tried analyzing 10 modes and qutip returned and overflow error, and I think switching to numpy might help solve this problem. And as a bonus there won't be a dependency on the qutip package :)

All of the uses of qutip are one of these:
* `qutip.qeye` ---> `np.qeye`
* `qutip.tensor` ---> `np.kron`
* `qutip.destroy/create/num/basis` ---> can be done in 3 lines with numpy, we can add this as a function in toolbox
* Replacing matrix multiplication `*` in qutip with `@` in numpy

I might do it myself later today, tell me what you think.

zlatko-minev, 2020-05-06:
It’s a good idea. Try and see if it helps. Qutip uses some sparse matrices here and there but not sure if that will help us, actually probably not since H isn’t sparse. 

We can then remove Qutip and/or add an option to export to it, by converting np.ndarray to QObj in a function

But probably not even much needed

DanielCohenHillel, 2020-05-07:
I removed the need for `qutip` from the package. I've did some testing on it and it worked  but since this could really break the library if done incorrectly I've pushed it into a new branch `remove_qutip` until we do enough testing to merge with master. Please check that and see if it works for you.

zlatko-minev, 2020-05-07:
Yes, this is super important to get right. We need to check the Hamiltonians that come out, not just the freq values.

DanielCohenHillel, 2020-05-07:
Yeah we should probably write some unit tests so we can make sure everything is correct. This would also be nice in the future, so we can make changes without being afraid that something broke along the way.

zlatko-minev, 2020-05-07:
Supplying those functions for H with some fake zero point fluctuations and checking the results against the hard-codes  numbers from the current Qutip code should work as a unit test