zlatko-minev/pyEPR #38: Add pyEPR to the Anaconda cloud

DanielCohenHillel, 2020-05-09:
Right now to install the pyEPR package you need to clone it and add it to the environment variables. I think it would be much simpler if we add pyEPR to the Anaconda cloud, this way to download the package all you need is to run something like `conda install -c minev pyEPR` and you're done. The upload process isn't that long and all you need is a free anaconda account. [Here is the documentation on how to do so](https://docs.anaconda.com/anaconda-cloud/user-guide/getting-started/#building-and-uploading-packages)

zlatko-minev, 2020-05-09:
Ok, I'll set up an account on anaconda.

zlatko-minev, 2020-05-20:
Ive been trying to do it like this with meta.yaml but right now the data isnt't even finding the version  ... 
```
{% set data = load_setup_py_data() %}

package:
  name: pyepr-quantum
  version: {{ data['version'] }}

build:
  number: 0
  script: python setup.py install --single-version-externally-managed --record=record.txt

requirements:
  build:
    - python
    - setuptools
  requirements:
    run:
      {% for req in data.get('install_requires', []) %}
        - {{ req }}
      {% endfor %}

test:
  imports:
    pyEPR

about:
  home: {{ data['url'] }}
  license: {{ data['license'] }}
  license_file: LICENSE
  summary: {{ data['description'] }}
  description: >
    pyEPR is an open source, BSD-licensed library providing high-efficiency,
    easy-to-use analysis functions and automation for the design of quantum
    chips based on superconducting quantum circuits, both distributed and lumped.
    pyEPR interfaces the classical distributed microwave analysis with that of 
    quantum structures and Hamiltonians. It is chiefly based on the energy participation 
    ratio approach; however, it has since v0.4 extended to cover a broad range of
    design approaches. pyEPR stradels the analysis from Maxwell’s to Schrodinger’s
    equations, and converts the solutions of distributed microwve (typically eignmode
    simulations) to a fully diagonalized spectrum of the energy levels, couplings,
    and key parameters of a many-body quantum Hamiltonian.
  doc_url: https://pyepr-docs.readthedocs.io/en/latest/
  dev_url: https://github.com/zlatko-minev/pyEPR

```

DanielCohenHillel, 2020-05-20:
I never really done something like that (except for the example they provide) so I can't really help you (managing packages/versions is always the annoying part :( ). Do you need to run setup.py first or something maybe? By the way, I'm guessing that you're trying to do this from your personal computer (a mac) and not a computer that actually runs the code (only works on windows :( ), since the package is only built for osx (you can see that on the [package page](https://anaconda.org/zlatko-minev/pyepr-quantum)). This means you can't install the package through conda unless you have a mac (and I don't have) so I can't install the package (using conda install simply returns that no such package exists).

zlatko-minev, 2020-05-20:
It is a mac. Hmm, i need to figure out how to get more channels. 

I fixed the data.get('version') issue.

Now i have some other issues with can't find setup.py ...

```
{% set data = load_setup_py_data() %}

package:
  name: pyepr-quantum
  version: {{ data.get('version') }}

build:
  number: 0
  script: python setup.py install --single-version-externally-managed --record=record.txt

requirements:
  build: 
    - python >=3.6
    - pip
    - setuptools
    - numpy >=1.15
    - attrdict
  run:
    - python >=3.6
    - pip
    - setuptools
    - numpy >=1.15
    - attrdict

test:
  imports:
    pyEPR

about:
  home: {{ data.get('url') }}
  license: {{ data.get('license') }}
  license_file: LICENSE
  summary: {{ data.get('description') }}
  description: >
    pyEPR is an open source, BSD-licensed library providing high-efficiency,
    easy-to-use analysis functions and automation for the design of quantum
    chips based on superconducting quantum circuits, both distributed and lumped.
    pyEPR interfaces the classical distributed microwave analysis with that of 
    quantum structures and Hamiltonians. It is chiefly based on the energy participation 
    ratio approach; however, it has since v0.4 extended to cover a broad range of
    design approaches. pyEPR stradels the analysis from Maxwell’s to Schrodinger’s
    equations, and converts the solutions of distributed microwve (typically eignmode
    simulations) to a fully diagonalized spectrum of the energy levels, couplings,
    and key parameters of a many-body quantum Hamiltonian.
  doc_url: https://pyepr-docs.readthedocs.io/en/latest/
  dev_url: https://github.com/zlatko-minev/pyEPR

```

DanielCohenHillel, 2020-05-20:
I really don't know nothing about it... they have a much simpler meta.yaml on the [example](https://github.com/Anaconda-Platform/anaconda-client/tree/master/example-packages/conda) that doesn't reference setup.py at all, maybe it's possible like that?

zlatko-minev, 2020-05-20:
Ok, youre right , i think i was biting off too much at once. I dumbed it down a bit without the setup info and now i can upload again.

DanielCohenHillel, 2020-05-20:
Cool :), don't forget to upload it from a windows computer as well.

zlatko-minev, 2020-05-20:
Figuring it out ... By the way, if you have a conda cloud account [I can add you](https://anaconda.org/zlatko-minev/pyepr-quantum/settings/collaborators)

DanielCohenHillel, 2020-05-20:
Thanks :smiley: I do have an account, the username is [DnL](https://anaconda.org/dnl). I can add a linux package although it wouldn't be any useful right now.

zlatko-minev, 2020-05-21:
I added you. Try this now: https://anaconda.org/zlatko-minev/pyepr-quantum

DanielCohenHillel, 2020-05-21:
I tried installing it but it didn't work, on windows I got this message:
```
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

The following specifications were found to be incompatible with your CUDA driver:

  - feature:/win-64::__cuda==10.1=0
  - feature:|@/win-64::__cuda==10.1=0

Your installed CUDA driver is: 10.1
```
I don't get what CUDA has to do anything but ok. I'll try checking what causing the issue, probably some of the dependencies are in conflict, maybe try to remove them for now only to check if they cause the issue?

Also tried it on a WSL and got:
```
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions
```

zlatko-minev, 2020-05-21:
Works for me
[Can you do a new conda env](https://docs.conda.io/projects/conda/en/4.6.0/_downloads/52a95608c49671267e40c689e0bc00ca/conda-cheatsheet.pdf)

conda create --name py36 python=3.6
conda activate py36
conda install -c zlatko-minev -c conda-forge pyepr-quantum

zlatko-minev, 2020-06-03:
I think this maybe done now. Both PyPi and conda forge, thanks to @willsALMANJ 

Try using: https://pypi.org/project/pyEPR-quantum/0.8/

and conda search -c conda-forge pyepr-quantum

zlatko-minev, 2020-06-03:
Could you test to see if these work for you? Thank you!

DanielCohenHillel, 2020-06-03:
Nice!! I will check it out

zlatko-minev, 2020-06-10:
Super! Glad to hear it seems to work. I will add somewhere in the docs soon how to upload and push changes to pypi and conda.

nathanshammah, 2020-09-16:
Maybe this issue could be closed? I was able to install it from conda (on Mac) with
```
conda install -c conda-forge pyepr-quantum
```
Another issue could be opened about documenting the installation in the docs. I am happy to help there.

zlatko-minev, 2020-09-16:
Yes, thank you. Let me close this issue then. Would you mind helping then with the doc part?

closed 2020-09-16 by zlatko-minev

nathanshammah, 2020-09-16:
Will be happy to.

zlatko-minev, 2020-09-16:
Awesome!