zlatko-minev/pyEPR #45: pyEPR: Working group meeting -- Getting together

zlatko-minev, 2020-08-13:
Let's get together for a pyEPR working group meeting. 

As pyEPR is growing a lot and being used by many labs and companies globally, it would be great to meet and chat about the future of pyEPR. 

Also, had a great chat with Unitary Fund about pyEPR for quantum design, so some folks from there may join us as well. 

**What do you guys think?** 

I can't seem to tag too many people, so may have to help spread the word. 

Maybe we can plan something in the next 4-6 weeks that works for the US and Europe timezones (any others I am missing?)? You can also e-mail me here [zlatko-minev.com](www.zlatko-minev.com)

DanielCohenHillel, 2020-08-14:
Sounds cool! Been a while since I contributed, should get back on it :smile: I'll look around the code to see what  I think should be changed

rlescanne, 2020-08-14:
Hi everyone, 
I would be glad to discuss with you guys. For the next 2 weeks, I will be on vacation but after that I'm fairly available.
Best,
Raphaël

zlatko-minev, 2020-08-14:
Great to hear, @DanielCohenHillel  and @rlescanne  (@leghtas). I have a few more responses in my inbox. Let's plan something out for a month to two months from now, how about mid-late Sept.? or can do second half of Oct. Let me know if you have any preferences. Would e-mail or here be best to organize?

rlescanne, 2020-08-19:
Mid-late september works for me !

leghtas, 2020-08-21:
Hi @zlatko-minev ! It would be great to discuss, sure mid-late September works for me too.

hodgestar, 2020-08-27:
@zlatko-minev I would love to come along and listen!

asafdi, 2020-09-01:
Any updates regarding the date of the meeting?

wslu42, 2020-09-10:
Hey @zlatko-minev this is Wen-Sen and I was working on the junction-embedded resonator which once used pyEPR. I'd love to sit in and listen if there's a remote meeting as well. Thanks!

zlatko-minev, 2020-09-11:
Alright folks, do you mind filling out this short form here, I will use the contact there send out the invite

https://forms.gle/g6nq3PaqP9835zDa6

[image]

zlatko-minev, 2020-09-11:
**[‼️ ‼️ MEETING SIGNUP -- Click link HERE ‼️ ‼️](https://forms.gle/g6nq3PaqP9835zDa6)**

ratangd, 2020-09-13:
Filled the scheduled meeting form as we discussed

zlatko-minev, 2020-09-14:
Great to hear

nathanshammah, 2020-09-14:
Glad to participate from the Unitary Fund side.

zlatko-minev, 2020-09-16:
Pleasantly surprised by the large number of people already signed up for #pyEPR I'm hosting-- 75!! 

Thank you everyone for spreading the word and singing up!

#quantumcomputing physics

[image]

zlatko-minev, 2020-10-23:
Excited to share the notes from the first #pyEPR working group meeting! We had between 30 and 40 folks on the call and stayed for an extra half hour, so thank you everyone for being great! 

**NOTES:** [2020-10 pyEPR Working Group Meeting 1 - Minev.pdf](https://github.com/zlatko-minev/pyEPR/files/5431670/2020-10.pyEPR.Working.Group.Meeting.1.-.Minev.pdf)

We will host another meeting in the next ~1-2 months and can focus more to let folks present what they have done. 

Key screenshots:

[image]
[image]
[image]
[image]

zlatko-minev, 2026-05-17:
Old working-group coordination from 2020. Closing as stale.

---
_Generated by [Claude Code](https://claude.ai/code)_