zlatko-minev/pyEPR #48: Get variable value from HFSS model

vegaonline, 2020-08-21:
The names of the variables cold be obtained from `pinfo.get_all_variables_names()`  as set of strings. It would be nice to see a function like  `pinfo.getValue('var name')` to return the value(s) of the variable name(s) of the HFSS model obtained from the previous command. This could be something like functions get() and set() used in C++ sort of classes.

zlatko-minev, 2020-09-11:
Not sure I completely follow. You can post minimal working code if you want. 

We will have a working group meeting for pyEPR soon:
 
Meeting signup link: https://forms.gle/oYsCjCMZLZtfrGuFA
Discussion link: https://github.com/zlatko-minev/pyEPR/issues/45
 
You can you sign up / help spread the word to anyone you think would be interest.

zlatko-minev, 2026-05-17:
Partially addressed in PR #191 (merged) — `pinfo.get_variable_value(name)` is now a first-class shortcut on `ProjectInfo`. Note it only reads local design variables, not global `$`-prefixed ones.

---
_Generated by [Claude Code](https://claude.ai/code)_