zlatko-minev/pyEPR #57: EPR on linux

zlatko-minev, 2020-10-23:
- [ ] There is a need to run on Linux and HPC cluster 

Nick Materise  will add notes here on what he has tried.

nmaterise, 2020-10-23:
Here is a summary of the things I tried to make pyEPR work on Linux
- The main issue is the absence of win32com on Linux
- There is a call to the Dispatch() method in pyEPR/pyEPR/ansys.py that returns the _app Application object in the HfssApp class. Dispatch() is part of the win32com module and is not available on Linux.
- There is a setup Python script in the Linux version of HFSS that has returns the oApplication and oDesktop objects needed to interact with HFSS, but they are separate objects that do not seem related in the same way as the Windows version. These objects are available in the IronPython terminal and scripting tool in Linux and you can export the start-up commands when you record and save an IronPython macro.
- I had to stop there and switch to the Windows version to try and make progress on pyEPR calculations before investigating further. It seemed doable to call or import the setup script and add the appropriate replacements for win32com calls, but I felt it was a rabbit hole at the time and did not pursue it further.

sergdanilin, 2023-11-30:
Thank you @nmaterise,

I am going through the same issue.
In the code base there is a comment

```
try:
    # TODO: Replace `win32com` with Linux compatible package.
    # See Ansys python files in IronPython internal.
    from win32com.client import Dispatch, CDispatch
except (ImportError, ModuleNotFoundError):
    pass #raise NameError ("win32com module not installed. Please install.")
```

Are there any updates on this ?

gsteele13, 2024-11-06:
Would it not be possible to interface to Ansys EDT using PyAEDT? 

https://github.com/ansys/pyaedt

I know that most of the rest of the qiskit metal code base has moved away from using win32 COM calls to using the official ansys interfacing API via PyAEDT. In fact, I think the only dependence left in qiskit metal that prevents it from running on linux is PyEPR?

zlatko-minev, 2024-11-07:
That’s right That would be great indeed Any chance anyone here wants to help set this up with some guidance and input from me?Sent from my iPhone

https://github.com/ansys/pyaedt

zlatko-minev, 2026-05-17:
Status update: the quantum analysis layer (loading HDF5, Hamiltonian diagonalization) works fully on Linux — no HFSS required. See the [without-HFSS guide](https://pyepr-quantum.readthedocs.io/en/latest/without_hfss.html). The COM interface for live HFSS control requires Windows. There is ongoing interest in [Palace](https://github.com/awslabs/palace) and PyAEDT-on-Linux as future paths for Linux EPR extraction. Keeping open as a roadmap item.

---
_Generated by [Claude Code](https://claude.ai/code)_