zlatko-minev/pyEPR #58: Adding linear resistance

zlatko-minev, 2020-10-23:
- [ ] Cap is experimental feature now (to be tested). Cj is set to a fixed value right now
- Adding in resistance calculation in a more unified way

Nick Materise

nmaterise, 2020-10-23:
I tried to hack the pyEPR/core_distributed_analysis.py file to compute the current across an external resistor as a crude attempt to model an RSCJ Josephson junction with a nonzero normal resistance.
- To do this, I modified ```get_junctions_L_and_C``` to return ```Ljs```, ```Cjs```, and another variable ```Rjs```
- Then, I modified ```calc_p_junctions``` to take ```Rjs``` as another input and called ```calculate_current_line_voltage``` with ```Rjs``` as another input
- Assuming that the junction is a parallel combination of ```Rj```, ```Lj```, ```Cj```, then the current becomes the ratio of the voltage V and the impedance Z, i.e. the code below
```python
722 omega = 2*np.pi*freq  # in SI radian Hz units
723
724        Z = omega*junc_L_Henries
725        if abs(float(Cj_Farads)) > 1E-29:  # zero
726            #print('Non-zero Cj used in calc_current_using_line_voltage')
727            #Z += 1./(omega*Cj_Farads)
728            print(
729                '\t\t'f'Energy fraction (Lj over Lj&Cj)= {100./(1.+omega**2 *Cj_Farads*junc_L_Henries):.2f}%')
730            # f'Z_L= {omega*junc_L_Henries:.1f} Ohms Z_C= {1./(omega*Cj_Farads):.1f} Ohms')
731
732        I_peak = V/Z  # I=V/(wL)s
```

could be replaced by something like this

```python
722        omega = 2*np.pi*freq  # in SI radian Hz units
723        if (junc_L_Henries > 0) and (Rj_Ohms > 0) and (Cj_farads > 0):
724            ZLR = 1./(1./(omega * junc_L_Henries) + 1./Rj_Ohms + omega * Cj_farads)
725        elif (junc_L_Henries > 0):
726            ZLR = omega * junc_L_Henries
727        elif (Rj_Ohms > 0):
728            ZLR = Rj_Ohms
729        else:
730            ZLR = 0.
731        Z = ZLR
732        if abs(float(Cj_Farads)) > 1E-29 & abs(float(junc_L_Henries)) > 1E-29:  # zero
733            #print('Non-zero Cj used in calc_current_using_line_voltage')
734            #Z += 1./(omega*Cj_Farads)
735            print(
736                '\t\t'f'Energy fraction (Lj over Lj&Cj)= {100./(1.+omega**2 *Cj_Farads*junc_L_Henries):.2f}%')
737            # f'Z_L= {omega*junc_L_Henries:.1f} Ohms Z_C= {1./(omega*Cj_Farads):.1f} Ohms')
738
739        I_peak = V/Z if (Z > 0) else 0. # I=V/(wL)s
```
Notes on the experimental capacitance feature: It is set to a constant (2 fF) in ```master``` pyEPR/core_distributed:
```python
1075 Cjs[junc_name] = 2E-15  # _parse(
1076                # 'Cj_variable') if 'Cj_variable' in val else 0
```

zlatko-minev, 2020-10-23:
Yes,`Cjs[junc_name] = 2E-15`  should be an optional property of the junction properties, I agree. 

The voltage is in parallel, so all the voltages are equal across RLC, but if there is an R, the R will such some of the energy and current out