zlatko-minev/pyEPR #60: Speed up transmons & incorporate new qubits (Agustin / Jose / Abhijit / Ratan)

zlatko-minev, 2020-10-23:
Based on pyEPR working group meeting

ratangd, 2020-10-23:
Hi ratan here

zlatko-minev, 2020-10-23:
Hi Ratan, thank you, let’s see what the next steps could be

vegaonline, 2020-10-24:
Hi,
this is Abhijit. It would be better if we have a discussion on what we
would like to envisage for the new qubits and transmon  speedup.

Abhijit Bhattacharyya

zlatko-minev, 2020-10-26:
Hi @vegaonline , do you want to start us off?

I guess fluxonium is one thing.

The other is the ability to put in any potential one would want

vegaonline, 2020-10-26:
> Hi @vegaonline , do you want to start us off?
> 
> I guess fluxonium is one thing.
> 
> The other is the ability to put in any potential one would want

Yes, these two topics are interesting.

zlatko-minev, 2026-05-17:
Still a long-term goal. The EPR approach is general and applies to any nonlinear element — transmon variants (fluxonium, gatemon, etc.) are natural extensions. The `calcs/` module provides the mathematical foundations. Contributions for specific qubit types are welcome. Keeping open as a roadmap item.

---
_Generated by [Claude Code](https://claude.ai/code)_