zlatko-minev/pyEPR #61: How do I contribute (Nick)

zlatko-minev, 2020-10-23:
[image]

nmaterise, 2020-10-23:
Thanks Zlatko, I will follow up here with more information

nmaterise, 2020-10-23:
Here is what I had in mind:
- Adding a section to the read-the-docs that covers how to contribute that would include the following
   - It would ideally point to the tests/ folder and encourage developers to test their code against these examples
      or give instructions on how to develop their own tests of their features
   - I found this blog post by dask developer Matthew Rocklin to cover best practices in open source contributions:
       http://matthewrocklin.com/blog/work/2018/10/12/so-you-want-to-contribute-to-open-source
   - This link has great advice for commit messages: https://chris.beams.io/posts/git-commit/
   - Other best practices in pull requests like squashing commits into a single commit for easier maintenance of the project

nmaterise, 2020-10-23:
I just checked out the README in the docs/ folder.
I could spend some time adding these to the docs as a separate pull request if there is broad interest in doing so.

zlatko-minev, 2020-10-23:
Excellent, thank you, @nmaterise. Maybe at the next meeting you can present these that you will add?

Zlatko

nmaterise, 2020-10-23:
Sure thing. I will see what I can put together by then. Thanks for organizing today's meeting. I hope can help to improve pyEPR and I am really excited to help with the other issues we started during the meeting.

zlatko-minev, 2020-10-23:
Thank you, @nmaterise ! Looking forward to it. You can post here if you need help with getting anything set up or feel like you need more direction. @DanielCohenHillel  has also been very active

zlatko-minev, 2020-10-23:
Thank you, Nick!

zlatko-minev, 2026-05-17:
Old internal coordination note from 2020. Closing as stale.

---
_Generated by [Claude Code](https://claude.ai/code)_