zlatko-minev/pyEPR #89: Problem when analyzing specific modes in QuantumAnalysis.analyze_variation

hatlabcz, 2021-12-09:
Hello,

I'm trying to do the EPR analysis for a multi-mode system, and it seems that analyzing more than 4 modes at once will take a very long time on my computer. So I tried to analyze fewer modes each time, say only mode 0 and mode 4, by passing [0, 4] to the 'modes' argument of QuantumAnalysis.analyze_variation. However, it seems that this will only analyze mode 0 and 1, not 0 and 4 as I wanted. 

I checked the source code and I think the problem is below:

https://github.com/zlatko-minev/pyEPR/blob/d507810f181446b862802c553bf51fa8775d20e6/pyEPR/core_quantum_analysis.py#L670 This line should be removed since we've already picked the mode frequencies in line 657

https://github.com/zlatko-minev/pyEPR/blob/d507810f181446b862802c553bf51fa8775d20e6/pyEPR/core_quantum_analysis.py#L671 
here (and below) the matrix index should just be [modes, :], since we want to analyze the modes specified in the argument, not the first len(modes) number of modes.

I can submit a pull request later that fixes this issue if these sounds correct to you.

Thanks.

zlatko-minev, 2021-12-13:
Thanks a lot for the issue.

That's interesting about the lines.

In general, when things move slowly, and you have many modes, usually it's because the numerical diagonal is being used and there are many many entries in the matrices.

Usually I will reduce the number of fock levels in that case.

Ideally, one could specify the number of truncation levels per mode, this is easy to write, it's not a lot of work, but it just hasn't been done yet. Very happy if you want to also look at this and make a pull request.

Regarding the two lines you mentioned above, this is a good idea. I'm not quite sure, because I have to really test and look at the code, but if you would like to make the change, test it on your end, see that it works, and then you can make a pool request, I'd be very happy to give feedback

zlatko-minev, 2026-05-17:
This is the same mode-selection behaviour documented in #161 (closed) and the [troubleshooting guide](https://pyepr-quantum.readthedocs.io/en/latest/troubleshooting.html). When you pass `modes=[0, 4]` to `do_EPR_analysis`, the EPR data is indexed by those original mode numbers. You must pass the same mode list to `analyze_all_variations` — passing no modes will attempt to look up index `[0, 1, 2, ...]` which won't exist. Closing as a duplicate usage question.

---
_Generated by [Claude Code](https://claude.ai/code)_