zlatko-minev/pyEPR #106: Running pyEPR with python 3.10

GyeonghunKim, 2022-03-31:
While running pyepr with python 3.10, importing ImportError occurred related to attrdict while importing pyepr. 

My code was just "import pyEPR as epr" and I attached an error message below. 

This problem is mainly about the syntax 
`from collections import Mapping`
is deprecated in python 3.10 and replaced to 
`from collections.abc import Mapping`.

While searching in Google, I found some news about attrdict is no more maintained, and I think this means we cannot expect whether they would fix this issue with python 3.10. Is there any plan to replace the attrdict library with another library or solve this issue differently?

Thanks. 

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File ~\Anaconda3\envs\pyepr_attrdict_test\lib\site-packages\pyEPR\__init__.py:78, in <module>
     77 try:
---> 78     from attrdict import AttrDict as Dict
     79 except (ImportError, ModuleNotFoundError):

File ~\Anaconda3\envs\pyepr_attrdict_test\lib\site-packages\attrdict\__init__.py:5, in <module>
      1 """
      2 attrdict contains several mapping objects that allow access to their
      3 keys as attributes.
      4 """
----> 5 from attrdict.mapping import AttrMap
      6 from attrdict.dictionary import AttrDict

File ~\Anaconda3\envs\pyepr_attrdict_test\lib\site-packages\attrdict\mapping.py:4, in <module>
      1 """
      2 An implementation of MutableAttr.
      3 """
----> 4 from collections import Mapping
      6 import six

ImportError: cannot import name 'Mapping' from 'collections' (C:\Users\aass98998\Anaconda3\envs\pyepr_attrdict_test\lib\collections\__init__.py)

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
Input In [3], in <cell line: 1>()
----> 1 import pyEPR as epr

File ~\Anaconda3\envs\pyepr_attrdict_test\lib\site-packages\pyEPR\__init__.py:80, in <module>
     78     from attrdict import AttrDict as Dict
     79 except (ImportError, ModuleNotFoundError):
---> 80     raise ImportError("""Please install python package `AttrDict`.
     81     AttrDict is in PyPI, so it can be installed directly
     82     (https://github.com/bcj/AttrDict) using:
     83         $ pip install attrdict""")
     85 ##############################################################################
     86 # Python header
     88 __author__ = "Zlatko Minev, Zaki Leghas, and the pyEPR team"

ImportError: Please install python package `AttrDict`.
    AttrDict is in PyPI, so it can be installed directly
    (https://github.com/bcj/AttrDict) using:
        $ pip install attrdict

GyeonghunKim, 2022-03-31:
I would like to work on this issue. Can you assign me to this issue?

WhiteSymmetry, 2022-04-01:
After implementing these solutions in Windows10&Python10.2.4, Qiskit-Metal started working:
1. anaconda3\envs\***\Lib\site-packages\attrdict (collections --> collections.abc) [corrected module: https://github.com/qfizik/AttrDict]
2. pyEPR (anaconda3\envs\***\Lib\site-packages\pyEPR\__init__.py: try: --> 
import collections.abc
try:
    collections = collections.abc
    from attrdict.dictionary import AttrDict as Dict
except AttributeError:
    collections = collections
)
[corrected module: https://github.com/qfizik/pyEPR/blob/master/pyEPR/__init__.py]
3.  Quantum-Metal 
(anaconda3\envs\***\Lib\site-packages\qiskit_metal\_gui\widgets\create_component_window\parameter_entry_window.py: 
from collections import OrderedDict, Callable --> 
from collections import OrderedDict
from collections.abc import Callable
)
[corrected module: https://github.com/qfizik/qiskit-metal/blob/main/qiskit_metal/_gui/widgets/create_component_window/parameter_entry_window.py]

[image]

[image]

GyeonghunKim, 2022-04-21:
@WhiteSymmetry Thanks for your comment. This method definitely works for individual users. However, without solving this problem inside pyEPR, every user should do that procedure every time they install the pyEPR and pyEPR-related libraries (such as qiskit-metal). Also, I think this is not appropriate for the CI/CD process.

closed 2026-05-17 by zlatko-minev