zlatko-minev/pyEPR #115: importing pyepr into metal during CI on Metal is failing.

priti-ashvin-shah-ibm, 2022-05-26:
When metal is using the new tag for pip, CI fails on Metal.
https://github.com/Qiskit/qiskit-metal/runs/6617275435?check_suite_focus=true

In vscode, one can see the error which would happen when imported using pip
[image]

﻿This is from https://www.geeksforgeeks.org/default-arguments-in-python/
[image]

@nikosavola , would you like to update the syntax, or I will try to do it tomorrow and have you be the reviewer?

nikosavola, 2022-05-27:
You can update the syntax 👍

nikosavola, 2022-06-04:
Should this issue be closed as of #116 ?

zlatko-minev, 2026-05-17:
This was fixed in 0.8.5.4 (2022) — the default-argument syntax issue was resolved and Metal CI has been green against pyEPR since then. Closing as stale.

---
_Generated by [Claude Code](https://claude.ai/code)_