zlatko-minev/pyEPR #161: pyEPR issue with selction of modes

davidjhonsonn, 2024-09-19:
​

First of all, thank you very much for your tutorials in pyEPR, they saved me!

I'm a master student of Roma3, working at INFN for a design of two transmon qubits. 

The code I'm using is similar to this from Daniel Cohen Hillel : [🔗 Coupling between the readout resonator and transmon - Daniel Cohen Hillel ](https://danielcohenhillel.github.io/tutorials/pyEPR-4-coupling-analysis/), in which he selects only 2 of the 5 modes (otherwise the simulation is to slow).

But with the same code I got an error:  **'[1] not in index'** when I run this cell: 
          epra = epr.QuantumAnalysis(eprh.data_filename)
          epra.analyze_all_variations(cos_trunc = 8, fock_trunc = 15);

Probably is something related to the index, because after selecting the modes 0 and 4 it doesn't find anymore the modes 1 in the array. 

I tried also selecting the mode again in the analyze_all_variations but I got the error **"index 4 is out of bounds for axis 0 with size 2"**

zlatko-minev, 2024-09-19:
Not sure from the simple message. Depends on the file. 

Check out: https://github.com/zlatko-minev/pyEPR/blob/551ec7114e0197cdc817ab6f4db70bf3db8ea9ed/pyEPR/core_quantum_analysis.py#L239 

You can see wha the loading does. You can print some of the statements there or the full error stack, not sure where the error is.

For instance print
epra.n_modes
epra.variations 
epra.freqs_hfss
epra.PM

davidjhonsonn, 2024-09-20:
thank you for the code. 

For instance print
epra.n_modes = 5
epra.variations = 1
epra.freqs_hfss = gives me the a panda.series with the 5 frequencies
epra.PM = gives me the a panda.series with 5 numbers

If you need I can send the full output but I don't think the problem is there. Perhaps is something related to the old version of pandas (I'm using version 1.5.3).

The strange thing is that it works if I select the modes [0, 1, 2], otherwise not. for example, If I select the modes [0,2,3] it gives me an error that index 1 is missing. Now I'm trying to change the order of the modes, [0,2,4,3,1] and then select the first three. Anyway it's not easy, if you know how to do change the order of the modes please help me. For instance, after I changed the order of modes, freqs_hfss_GHz remains uncghanged. So I manually change it. But now f_ND results are unchanged, too.

zlatko-minev, 2024-09-20:
Yes, it could be the indexing. What's the print stack trace of the error / what is the line where this happens

davidjhonsonn, 2024-09-21:
I solved the problem changing the code of pyEPR\core_quantum_analysis, line 670, where it gives me the error of indexing. 
Let me know if my solution it's not a good one and if you have other idea.
Thank you for your attention.

I select the mode here:
modes=[1,2,4]
eprDist.modes = modes

var_select = None

data_file,var_return = eprDist.do_EPR_analysis(variations=var_select, modes=modes)

Then I have to run the quantum analysis.Here is the line: 
eprQ.analyze_all_variations(**Qanalysis_settings, variations=var_select, modes=modes)

The problem is that if I don't select the modes in the quantum analysis it dosen't recognize the index (for instance with modes [0, 2, 4] it says index 1 is missing. 

If I select the modes also here it says "index 4 is out of buonds for axis 0 with size 3". The problem is that the code uses the array modes to address the variables like PJ, SJ and so on. So, when the modes are [0, 2, 4] it try to access to the fifth element of PJ but it has only three elements (0, 2 and 4). 

So I changed the original code using an array of indices instead of the array of the modes for those variables:

[image]