zlatko-minev/pyEPR #162: incompatibility with numpy > 1.19? (AttributeError: 'complex' object has no attribute 'norm')

gsteele13, 2024-11-06:
In a recnet install, I ran into this error:

```
File ~\AppData\Local\miniforge3\envs\qiskit-metal-pip\lib\site-packages\pyEPR\calcs\back_box_numeric.py:225, in make_dispersive.<locals>.closest_state_to.<locals>.distance(s2)
    224 def distance(s2):
--> 225     return (s.dag() * s2[1]).norm()

AttributeError: 'complex' object has no attribute 'norm'
```

Some googling suggests that this is related to problems with changes in numpy between 1.19 and 1.20:

https://groups.google.com/g/qutip/c/R0kXhv1rtO8
https://www.reddit.com/r/QuantumComputing/comments/1dll5gh/pyepr_tutorial_1_questions/

It looks like a pretty simple fix? I may look into fixing it and submitting a pull request.

gsteele13, 2024-11-06:
ok, to my great sadness, reinstalling everything and downgrading to numpy 1.19 did not get my code to run. i will indeed have to pull the pyepr code and fix the bug.

gsteele13, 2024-11-06:
OK, I realise that I am too much of a noob to figure out how to create a pull request. But in any case, this diff fixed my problem:

[image]

I see .norm() used a few more times in the file, but I have not yet run into issues with it causing errors.

zlatko-minev, 2024-11-07:
Thank Gary! This is nice.You could use the online interface and just click the edit file icon at the top When you try to save it will ask if you want to set up as clone and pullSent from my iPhone

image.png (view on web)

gsteele13, 2024-11-07:
Hi Zlatko,

Yes, in fact, during the pip install ., it upgraded already numpy to 1.24. I also just did some testing and it works fine, this was my test:

```
mamba create -n qiskit-metal-pip python=3.10 pint=0.23
mamba activate qiskit-metal-pip
mamba install qiskit-metal notebook
pip install git+https://github.com/gsteele13/pyEPR.git
jupyter notebook
```

then Open 4.02 Eigenmode and EPR.ipynb => Make sure all cells run

Python 3.12 does not work with qiskit metal (kernel crashes on import) so I rolled back to 3.10 there. And, furthermore, pint 0.24 also breaks qiskit metal, I found that rolling back to 0.23 solves the issue. I should probably make some issues in the qiskit-metal repo (may do that when I have time...) 

In any case, with a pretty fresh install based on a constraint to python 3.10, the code in this pull request works fine.

zlatko-minev, 2024-11-08:
Thanks @gsteele13  
Yes we can update the requirements if we are confident what new ones will work

ball199578, 2024-11-22:
Thanks.I use Gary‘s to solve this problem and it works.But if numpy>=2.0(my numpy=2.1.3),I meet another problem:`np.mat` was removed in the NumPy 2.0 release. Use `np.asmatrix` instead.(core_quantum_analysis.py:788),my coding ability is very poor,cannot change it.So I change numpy2.1.3 to numpy1.25.It works and can run all the code of Startup example.

zlatko-minev, 2024-11-22:
Nice, so there is going to be a massive merge and update coming in from here: 
https://github.com/zlatko-minev/pyEPR/pull/153

It would be great to review and make sure your end works with it, since this pull will be the next version. Ideally we address and fold in the info from here there with your help